edaplayground/eda-playground #15: ERROR VCP2890 "Unspecified argument is used for an argument that does not have a default value: name." "testbench.sv" 15  2850

reddyyogesh, 2015-01-29:
i am getting the following error with Riviera-PRO EDU 2014.10, but same work with modelsim,
i can not use modelsim because of radomize() limitation.

please suggest how to fix this issue.

--------------------------------------------------------------------------------------------------------
ERROR VCP2890 "Unspecified argument is used for an argument that does not have a default value: name." "testbench.sv" 15  2850
-------------------------------------------------------------------------------------------------------
`include "uvm_macros.svh"
 import uvm_pkg::*;

class Packet extends uvm_transaction;
    rand bit [7:0] length;
    rand bit [7:0] da;
    rand bit [7:0] sa;
    rand int data[];

    `uvm_object_utils_begin(Packet)
       `uvm_field_int(da, UVM_ALL_ON|UVM_NOPACK)
       `uvm_field_int(sa, UVM_ALL_ON|UVM_NOPACK)
       `uvm_field_int(length, UVM_ALL_ON|UVM_NOPACK)
       `uvm_field_array_int(data, UVM_ALL_ON|UVM_NOPACK)
    `uvm_object_utils_end
  
      function new(string name = "");
         super.new(name);
    endfunction
endclass : Packet

/////////////////////////////////////////////////////////
////    Test to check the packet implementation      ////
/////////////////////////////////////////////////////////
module test;
    Packet pkt1 = new("pkt1");
    initial
    repeat(10)
//       if(pkt1.randomize)
//       begin
          $display(" Randomization Sucessesfull.");
         // pkt1.print();          
//       end
//       else
//       $display(" *** Randomization Failed ***");
endmodule

getvictor, 2015-01-29:
+Sunil from Aldec

Sunil,
Can you take a look at this issue? The code appears correct. Here is a
link: http://www.edaplayground.com/x/9yS

Thanks,
Victor

getvictor, 2015-01-30:
---------- Forwarded message ----------
From: Sunil Sahoo <sunils@aldec.com>
Date: Fri, Jan 30, 2015 at 10:49 AM
Subject: RE: [eda-playground] ERROR VCP2890 "Unspecified argument is used
for an argument that does not have a default value: name." "testbench.sv"
15 2850 (#15)
To: Victor Lyuboslavsky <victor.lyuboslavsky@gmail.com>

Yes,

The UVM in the Riviera-Distribution is slightly modified to work with
Riviera. It is modified in such a way that the if any other simulator uses
the Riviera UVM library, it will still see the Accellera version. But we
have IFDEF’s in the code which see if Riviera PRO is the simulator, then
the modified code is used.

Sunil

*From:* Victor Lyuboslavsky [mailto:victor.lyuboslavsky@gmail.com]
*Sent:* Thursday, January 29, 2015 8:27 PM
*To:* Sunil Sahoo

*Subject:* Re: [eda-playground] ERROR VCP2890 "Unspecified argument is used
for an argument that does not have a default value: name." "testbench.sv"
15 2850 (#15)

Sunil,

The UVM is the standard distribution from Accellera. So the one with the
install has been modified to work with Riviera-PRO?

Interestingly, UVM 1.1d seems to work fine with the above example. It is
only UVM 1.2 that is hitting this issue.

-Victor

On Thu, Jan 29, 2015 at 6:08 PM, Sunil Sahoo <sunils@aldec.com> wrote:

Victor,

I tried the same code on my local machine and it compiles without errors.
Can you check if the $UVM_HOME is pointing to the correct UVM directory
inside the riviera installation.

Regards,

Sunil

*From:* Victor Lyuboslavsky [mailto:victor.lyuboslavsky@gmail.com]
*Sent:* Thursday, January 29, 2015 5:01 AM
*To:* getvictor/eda-playground; Sunil Sahoo
*Cc:* getvictor/eda-playground
*Subject:* Re: [eda-playground] ERROR VCP2890 "Unspecified argument is used
for an argument that does not have a default value: name." "testbench.sv"
15 2850 (#15)

+Sunil from Aldec

Sunil,

Can you take a look at this issue? The code appears correct. Here is a
link: http://www.edaplayground.com/x/9yS

Thanks,
Victor

<https://github.com/getvictor/eda-playground/issues/15>.[image: Image
removed by sender.]

reddyyogesh, 2015-01-30:
Thanks for spending time for this issue.
So what is the final conclusion, do we have to use UVM 1.1d till this is fixed in UVM 1.2

Thanks
Yogesh