edaplayground/eda-playground #16: Can't logout, then can't log-in anymore

ReubenMijares, 2016-09-21:
I cannot logout my account. Every time I go to EDA Playground, my name is shown on the upper right side corner. When I click it, the page will load for a while but it goes to the same page where I'm currently in. I can't logout my account. I can't run any simulations too.

This happens after I clicked the validate email link from my Gmail account. This is not the first time I clicked that link. I have already validated my account sometime ago. I just clicked that to see if it will take me directly to the EDA Playground website.. It did, but now I can't logout.

Please help.

mpataylor, 2016-09-21:
Hi Reuben,

Sorry you're having trouble. This isn't an issue that I have come across before. My first suggestion would be to delete all the EDA Playground cookies and see whether that fixes the issue. (In Firefox you would do that by selecting Tools -> Options -> Privacy -> remove individual cookies  and then enter "playground" in the search box to find the EDA Playground cookies, select them and click "Remove Selected". I guess there will be a similar procedure in some other browser.

Let me know how you get on,

Matthew

ReubenMijares, 2016-09-21:
Hi Matthew,

Thanks for your reply.
Anyway, the issue is already gone. I can now use EDA Playground again.
I didn't do anything. I just ate lunch then after I come back to my desk, it is now okay.

Anyway, thanks for giving another possible solution.

closed 2016-09-21 by ReubenMijares