edaplayground/eda-playground #17: Gate Level Simulations

ReubenMijares, 2016-09-21:
I would like to practice Gate Level Simulations. So I would like to request EDA Playground to have this feature. It's good way to experience how to debug setup and hold time violations, as well as x propagation issues.

Thanks in advance.

mpataylor, 2016-09-21:
Hello again Reuben,

At one level a gate-level simulation is no different to any one kind of simulation. If you loaded up you testbench, gate-level netlist and gate library into separate tabs, you could just run a simulation like any other. So, I'm a bit unsure as to what needs to be added to EDA Playground specifically for gate-level simulations. 

Matthew

ReubenMijares, 2016-09-21:
Hi Matthew,

I see. I think there is something wrong with what I'm doing that's why I can't run GLS.
Actually, I'm looking for the gate-level netlist. I tried Yosis but it only gave me a schematic. I don't know where the netlist is generated.

ReubenMijares, 2016-09-21:
Hi,

I tried Yosys again and now I selected the option ABC cell library and download files.
I think the file synth.v is the one I'm looking for. Am I right?
Does this have any gate delays?

mpataylor, 2016-09-21:
I tried selecting the ABC cell library option and see that synth.v now refers to two gates ( NOT _0_ and DFFSR _1_ ) but to run a simulation we would need to have some code underneath both those. I don't know whether that is around or not - I would have to look into it.

So, some kind of option to simulate the output of Yosys would be interesting: I will add it to my ToDo list (but don't hold your breath).

So, I think it might be possible to run a gate-level simulation of the output of some other synthesiser (outside EDA Playground) by uploading the testbench, gate-level netlist and gate library. I guess you could probably use back-annotation to include delays from an SDF file (also uploaded). It would be an interesting example; if I get a chance, I'll try it.

ReubenMijares, 2016-09-22:
I see. I guess I need to try the other method you are saying.
Anyway, I'm looking forward on the option to allow simulation of the output of Yosys. I know it may take time. Thanks in advance. =)

closed 2016-09-22 by ReubenMijares