edaplayground/eda-playground #18: Add support for VUnit

LarsAsplund, 2016-10-30:
[VUnit](http://vunit.github.io/index.html) is an open source unit testing framework for VHDL and SystemVerilog proving the functionality needed to realize continuous and automated testing of your HDL code. 

VUnit is very simple to install (a one-liner if Python is installed) and try out but I'm looking for ways to demonstrate the tool which doesn't require any files to be installed. EDA playground seems like the perfect fit for that.

Full automation means going beyond writing self-checking testbenches. It also includes automation of compilation and testbench execution for all testbenches in your project. This part is done with a [Python run script](http://vunit.github.io/user_guide.html#introduction). This means that an example on EDA playground would be at least three files - the testbench, the run script, and the design.

We support a number of different [simulators](http://vunit.github.io/about.html#Simulators). Two of them - Riviera-PRO and Cadence Incisive are supported by you as well. However, our Incisive support is experimental so I would suggest starting with Riviera-PRO.

Doable? I would be happy to help out in any way I can.

umarcor, 2020-04-01:
@edaplayground, any update on this issue?

edaplayground, 2020-04-06:
Yes. It's been on EDA Playground for quite a while.

umarcor, 2020-04-12:
@edaplayground, I don't know where I was looking to. I must be blind. Of course, I found it now.

May I suggest to:

- Have it shown as an option in https://www.edaplayground.com/playgrounds? Either as a 'Methodology' or (better) as a checkbox.
- Have the version of VUnit shown (I think that it is v3.3.0):

```py
from vunit.about import version
print(version())
```
- Have VUnit and GHDL updated. Latest stable releases are v4.4.0 and v0.37, respectively.

I'd expect that updating both VUnit and GHDL would make it possible to use them together.

LarsAsplund, 2020-05-05:
Now that Questa is available would it be possible to enable VUnit for that?

edaplayground, 2020-05-05:
I've put it on the ToDo list.