edaplayground/eda-playground #23: Error in execution

RobertCraven90, 2017-11-22:
EdaPlayground is not working at the moment.

It started a few hours ago, where the Homepage was temporarly unaccessible a few times and sometimes the Compiler did not start (I can't remeber the error message).

Now the following Error Message occures everytime I try to run any code:

Exit code expected: 0, received: 1
Error in execution: [No such image: 1511343044018-0 (tag: latest)
]. Your run may have reached the memory limit of 500MB.

ydonia, 2022-11-16:
I am having the same issue. No matter what code or language I am running, I get the following error with no other logs:
Exit code expected: 0, received: 1
Error in execution: [No such image: 1668631156903-0 (tag: latest)
]. Your run may have reached the memory limit of 500MB.

I would appreciate any help, I can't seem to find other resources online that help.

mpataylor, 2022-11-18:
One of the servers had got into a funny state. I rebooted it. It should be OK now. Sorry for the inconvenience.

closed 2022-11-18 by mpataylor

SrushtiPW, 2022-12-04:
Getting the error "You may have reached the memory limit of 500MB" no matter which code I try to run. Please fix this at the earliest. It would be much appreciated.