edaplayground/eda-playground #26: using ghdl simulation exits with empty design.vhd, testbench.vhd

MartinMosbeck, 2018-08-08:
Hi,

our students use edaplayground the following way:
* Add (Upload) a testbench file and some entities+architecture files to a playground
* Leave the default design.vhd and testbench.vhd empty
* Adjust the top entity field, run

We now want to use ghdl as the simulator, but ghdl exits when it given an empty file to analyze:
    `design.vhd:2:1: design file is empty (no design unit found)`
    `testbench.vhd:2:1: design file is empty (no design unit found)`
    `ghdl: importation has failed due to compilation error`

It is true that "The VHDL language does not allow empty files (including comment only files)."(https://github.com/ghdl/ghdl/issues/519) so it might make sense to stop the process with an error, neither the less other simulators handle this as a warning instead of an error.

I do not see a way to exclude the design.vhd and testbench.vhd from the simulation or delete them from the playground. Any ideas how I can solve this problem?

edaplayground, 2018-08-15:
Hi Martin, 

Sorry for the delay in replying - I was out of the office on holiday (not something I want to broadcast on the internet until I am back and my house occupied again).

I have come across this and there is no nice, easy fix for you. The not so nice, not so easy fix is to just put a dummy design unit in each file. 

I need to think about how this could be fixed in EDA Playground. Eg I could automatically exclude empty files. BUT a completely empty file is probably quite easy to detect, whereas a file that doesn't contain a VHDL design unit is much harder to detect. Another option would be to allow the deletion of the testbench.vhd and/or design.vhd tabs. 

Matthew

MartinMosbeck, 2018-08-15:
Hi Matthew,

thanks for the reply, hope your holiday was nice! 

The "not so nice, not so easy fix" isn't really good for my case, as we want to make things as easy for students as possible and not confuse them with workarounds they have to apply (and probably half of all students will forget and come with a "why is this not working" to us).

I personally think excluding completely empty files from the simulation is the best way. Additionally, I see no scenario where nonempty files without at least one design unit make much sense. If someone creates such a file it's likely that he/she forgot to add a unit in the file and therefore, in this case, the error message is appropriate. And detection of unitless files is done by ghdl anyway.

I don't know how you create the commands which are executed to interface with simulators, but I imagine adding a ignore totally empty files there has less probability to cause global bugs and is less work to implement than making testbench.vhd & design.vhd deletable.

Thanks for your effort!

Martin