edaplayground/eda-playground #27: EdaPlayground unstable or offline

RobertCraven90, 2018-10-16:
Hi!

EdaPlayground ist very unstable over the last few days :(
Website is often offline and the Simulation Runs do not start or produce an error.

Are there some maintenance works running?

best
RobertCraven

mpataylor, 2018-10-16:
Hi Robert,

 

Hasn't it? I'm sure it's very annoying. Sorry for the disruption.

 

The truth is I'm trying to find out what's wrong, but so far haven't managed to. It's related to usage - it's worse in the mornings (UK time) when India is making heavy use of it.

 

Best regards,

 

Matthew

__________________________

Matthew Taylor | Senior Member of Technical Staff 

matthew.taylor@doulos.com | Skype mpataylor  | www.doulos.com
T +44 1425 517633 | M +44 7789 748993

RobertCraven90, 2018-10-16:
Hi!

Thanks for the fast reply, hope that you will find a way :)

At the moment the web interface seems to be stable, but most of the time I get the error message:

Exit code expected: 0, received: 1
Error in execution: [No such image: 1539695487635-0 (tag: latest)
]. Your run may have reached the memory limit of 500MB.

when running VHDL Code.

best RobertCraven

closed 2024-09-12 by edaplayground