edaplayground/eda-playground #30: eda-playground down?

dregenold, 2019-02-07:
Trying to run some simulations which worked last night, but today I'm getting an error message which says, "Error: Failed to create temporary directory within 10000 attempts (tried /var/lib/tomcat7/temp/1549567448043-0 to 1549567448043-9999)"

tanayverma, 2019-02-07:
I am getting the same issue. From what I can see in the forums, this is an issue that last surfaced in 2017.

mpataylor, 2019-02-08:
This is now fixed. Sorry for the inconvenience.

suryakrish31, 2025-07-30:
Hi i am getting the same error now when trying to run the simulations.

suryakrish31, 2025-07-30:
Yes, it is 205. Then, is there any way, i can change it?

suryakrish31, 2025-07-30:
Yeah, I refreshed the paged and the number changed. Thank you for the support.

KienVu2002, 2025-07-30:
Hi, i got the same situation, my ID of the server is 205. How can i change it although i refreshed the page many times?

eileenrhickey, 2025-08-01:
Sorry - wrong button