edaplayground/eda-playground #31: Execution interrupted or reached maximum runtime.

Shahid0606, 2019-04-03:
EDA Playground was working fine for me yesterday but today I am continuously getting the following message for anything I do

Execution interrupted or reached maximum runtime.

I have tried pretty much everything like my own code, example codes etc but to no avail. Can you please help? Thanks

mpataylor, 2019-04-03:
Hi,

The maximum run time on EDA Playground is 1 minute. So, if your simulation is longer than that, or if it never stops, then it will time out. That is the meaning of your error message.

Matthew

Shahid0606, 2019-04-03:
Thanks Matthew 

I have tried the provided UVM/SV/Python etc examples as well besides my own test benches. Those examples as well as my TBs were working fine for me yesterday. To me there is some other problem. Can  you please help??

Shahid0606, 2019-04-03:
I am still stuck, don't know why :(. Very frustrating :(

mpataylor, 2019-04-03:
I can't see anything wrong. Please could you post a URL of a playground that is giving you trouble?

mpataylor, 2019-04-03:
I'm sorry. One of the servers had got into a funny state. It is now fixed.

Matthew

Shahid0606, 2019-04-04:
Thanks a lot Matthew!!

Prabhat-lab, 2021-05-26:
I am facing that EDA Playground problem again regarding that "Execution interrupted or reached maximum runtime".
And for each and every program, they were all working fine till yesterday, now none of them are getting executed.
Please help.

edaplayground, 2021-05-27:
One of the servers had got into a funny state. It should be fixed now. Sorry for the inconvenience.

closed 2021-05-27 by edaplayground

mykytko, 2021-10-05:
I am having the same issue with default examples, "Execution interrupted or reached maximum runtime". Please help.

mpataylor, 2021-10-05:
One of the servers got into a funny state. I rebooted it, but it has got back into the same state. I have taken it offline pending further investigation. Sorry for the inconvenience.

lucasazuma, 2021-10-05:
the same thing is happening to me now, any quick solves?

Pitica-Sebastian-Razvan, 2022-03-30:
Same problem "Execution interrupted or reached maximum runtime". Please help.

Pitica-Sebastian-Razvan, 2022-03-30:
@edaplayground @mpataylor

mpataylor, 2022-03-30:
It is the same thing. Sorry. I have removed the server from the pool. (I do have an item on my ToDo list to look into automatically detecting this, but it hasn't reached the top yet.)

Pitica-Sebastian-Razvan, 2022-03-30:
"Error connecting to server. Run request failed: error" at the moment. I should be ok later?

chiradeepdey, 2022-05-17:
I am facing the same issue "Execution interrupted or reached maximum runtime". Please help @mpataylor @edaplayground

mbhdra, 2022-05-25:
Same problem occurs now. Could you please help?

mpataylor, 2022-05-25:
It may be because EDA Playground was pretty busy. Or it may be because of a problem in your code, such as an infinite loop.

mpataylor, 2022-05-25:
One of the servers had got into a funny state. I had to do a power cycle (a reboot wasn't enough). Sorry for the inconvenience.

CHUNKUANCHIH, 2022-06-22:
My friend also got the same problem. My computer is working fine, but she doesn't. Could you help me with this issue? Thanks

youssefelbaz32, 2022-07-27:
Same problem has just happened again @mpataylor.

mpataylor, 2022-07-27:
This is fixed now.

PALLAVI-96, 2022-08-05:
Hi, "Execution interrupted or reached maximum runtime. This process not found on server." Getting this error multiple times today. Can you help @mpataylor

edaplayground, 2022-08-10:
Hello,

The error should be fixed now

nithinbhargav007, 2022-08-14:
I am facing the same issue, very frustrating.  Not able to run any code

averma0314, 2023-01-12:
Have been facing this issue since yesterday. Can you please look into it? @mpataylor @edaplayground

mpataylor, 2023-01-13:
Hopefully it s fixed now. Sorry for the inconvenience.

alamgirakash2000, 2023-02-02:
@mpataylor @edaplayground could you please fix this again? I am facing the same problem.

dokes288, 2023-02-20:
Hi...I am having the same error as observed on the Edaplayground page where I am designing  a 4bit LFSR with two inputs - XOR gates and a SEED of 1111. Here is the link: https://www.edaplayground.com/x/uUG3. However, I am new to Edaplayground and Verilog, and so I can't tell if the EPwave not showing the simulation is from the server or from my code.

Thank you,

Dominic

mpataylor, 2023-02-20:
You simulation never finishes. Therefore, EDA Playground is timing it out after 1 minute. 

The reason why it never finishes is because of this line:

      always #5 clk=~clk;

which just keeps on going forever. You either need to put this inside some kind of (eg) while loop, which can be disabled. Or you need to call $stop; eg:

    [https://www.edaplayground.com/x/As9W](https://www.edaplayground.com/x/As9W)

dokes288, 2023-02-25:
Thank you for the debugging. I  have generated the simulation.

Dominic Okeke

 
 

[https://www.edaplayground.com/x/As9W](https://www.edaplayground.com/x/As9W)

tobias1910, 2023-03-27:
why even this code isn't running?

program testbench();
  initial begin
    $display("Hello World!");
    #1;
  end
endprogram

I used Icarus 9.7 says this process is not found on server

edaplayground, 2023-03-27:
The "Process not found on server" is a bug, which I've not had time to fix. Try reloading the page. This code won't run n Icarus, because Icarus doesn't understand SystemVerilog. (Or at least this version doesn't.)

tobias1910, 2023-03-28:
Yeah.. It works in other compilers ..Thank a lot!

abcoadaoihc, 2023-04-27:
Hi
I am having the "execution interrupted or reached maximum runtime" error with my SystemVerilog code. I am designing a 1D convolutor with two 6-bit inputs, and here is the link to my code: https://edaplayground.com/x/mEaa. I don't see any infinite loops? But I am not really good at SystemVerilog so I may miss something...
Thanks a lot.

edaplayground, 2023-04-28:
Hi,

It looks to me like you've fixed it?

Best regards,

Matthew

yushuang-liu, 2023-05-01:
> Hi, It looks to me like you've fixed it? Best regards, Matthew
> […](#)

Hi, 

It was fixed, but it's happening again. Could you take a look at it please?

Thanks

edaplayground, 2023-05-03:
To be honest, I think it is just very heavy use on EDA Playground.

tiraoelmas, 2023-06-01:
Hi!, i think one of the servers fall into a funny state again bc i tried for the last two hours work with eda and the message "Execution interrupted or reached maximum runtime. This process not found on server." stills appeares, if u can fix it pls :c

mpataylor, 2023-06-01:
Please can you tell me the number at the bottom left-hand corner of the screen. (It will be in the range 160-175.)

Matthew

tiraoelmas, 2023-06-01:
its in 165, but i think you fixed the page bc now eda works perfectly, ty sm Matthew! 
Cristóbal

mpataylor, 2023-06-01:
I'm glad to hear it's working again, but I didn't do anything. It seems to have sorted itself out. I will have a look at some logs.

Matthew

Nemanja264, 2024-04-30:
Execution interrupted or reached maximum runtime.
This process not found on server.
i have this issue over and over, any code i run this error pops up.
yesterday it was doing fine, i tried everything, im sure that i dont have any infinite loops.
i reduced my runtime but still this error appears, can you please help me.
Here is my code:
-- Code your design here
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all ;

entity brojac is
	generic(n: positive:= 10);
    port(smer: in std_logic;
    	ce: in std_logic;
        wr: in std_logic;
        din: in positive;
        clk:in std_logic;
        counter: out positive
        );
end entity;

architecture arch of brojac is
begin
	per: process(clk,ce,wr)
    variable count:positive:=0;
	begin
    if clk'event and clk='1' then
    	if ce='1' then
        	if smer='1' then
            	if count=n-1 then
                	count:=din;
                 else 
                 	count:=(count+1) mod n;
                 end if;
              else
              	if count=0 then
                	count:=din;
                 else
                 	count:=(count-1) mod n;
                 end if;
             end if
        else if wr='1' then
        	count:=din;
        end if;
    end if;
    counter<=count;
    end process;
end architecture;

-- Code your testbench here
library IEEE;
use IEEE.numeric_std.all ;
use IEEE.std_logic_1164.all;

entity tb is
generic(n:positive := 10);
end tb ; 

architecture a_tb of tb is
    signal clk, wr, ce, smer: std_logic;
    signal counter, din : positive;
    
begin
    dut : entity work.brojac(arch)
    	generic map(n);
        port map(smer,ce,wr,din,clk, counter);
    
    clock: process
    begin
        clk <= '0'; wait for 5 ns;
        clk <= '1'; wait for 5 ns;
    end process;
    
    stimuli: process
    begin
        din <= 2;
        smer <= '1';
        wr<='1';
        ce<='0';
        wait for 10 ns;
    

    end process;

end architecture ;

mpataylor, 2024-04-30:
This process

clock: process

begin

    clk <= '0'; wait for 5 ns;

    clk <= '1'; wait for 5 ns;

end process;

never ends. Thus, it keeps generating "events" (a change on a signal) every 5ns. If there exist future events, a VHDL simulation will just keep running. One way of stopping a VHDL simulation is to stop the creation of events. (This is called "event starvation".)  Consequently, EDA Playground times your simulation out after 1 minute. Look into using a "while loop" to generate your clock.

BTW, EDA Playground is really good for code-sharing. You can share the code by simply sharing the URL. This makes it easy for someone else to help, because they can also run your code.)

Matthew

lmadem, 2024-05-01:
The issue "Execution interrupted or reached maximum runtime" is troubling me since few hours. Couldn't run even a small piece of code. Please help!

edaplayground, 2024-05-01:
> You don't often get email from ***@***.******@***.***>. Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>

Three of the servers were down for some reason. I have rebooted them and I think they are OK now. Sorry for the inconvenience.

Matthew

mpataylor, 2024-05-01:
Three of the servers were down for some reason. I have rebooted them and I think they are OK now. Sorry for the inconvenience.

Matthew

vcheung-radical, 2024-05-03:
Execution interrupted or reached maximum runtime.
This process not found on server.

Down again?

HopeHaz, 2024-05-06:
Even I am having this issue "Execution interrupted or reached maximum runtime." How to overcome this ? Please respond as soon as possible

lmadem, 2024-05-06:
It is working for me now. Thank you!

edaplayground, 2024-05-06:
It should now be fixed. I have been away for the weekend and today is a public holiday in the UK (I just got back).

oscarhsing, 2024-05-19:
excuse me. I have the question about eda playground. I have ran my playground. but I get the response of "Execution interrupted or reached maximum runtime." can you help me to solve that question? thank you.

oscarhsing

mpataylor, 2024-05-20:
One of the great things about EDA Playground is that it is possible to share code by sharing the URL of that code. This is useful when you are asking for someone's help. Please can you share the URL of your code (and make sure that it is public).

Matthew

oscarhsing, 2024-05-31:
here is my url.
https://www.edaplayground.com/x/RH2q

edaplayground, 2024-05-31:
As I suspect, the loop that generates your clock runs forever.

A VHDL simulation runs until

  *   the simulator stops it
  *   a VHDL-2008 STOP or FINISH procedure is executed or
  *   it runs out of things to do. To put this more formally, no more events are generated. (This is called event starvation.)

Your clock-generation process generates events (changes on a signal) forever and so the simulation runs forever. However, EDA Playground times it out after 1 minute.

One solution to this is at: https://www.edaplayground.com/x/RTKK .

Matthew

https://www.edaplayground.com/x/RH2q

-

oscarhsing, 2024-05-31:
hi, Matthew. can I ask you how to fix it. 

Oscar

oscarhsing, 2024-05-31:
thank you for your help.

Oscar

oscarhsing, 2024-06-01:
sorry, I have the question again. why my p_next cannot show the wave directly?
https://www.edaplayground.com/x/q6_f

mpataylor, 2024-06-03:
I can see p_next in the waveform.

https://www.edaplayground.com/x/q6_f

-