edaplayground/eda-playground #33: Add support for cocotb 1.0 and 1.1

eric-wieser, 2019-07-14:
Follow-up from #5.  There have been two new releases of cocotb since then, which are listed at https://github.com/potentialventures/cocotb/releases.

It would be great to add support for at least 1.1, and 1.0 if you think it is worth the time investment

edaplayground, 2019-08-07:
I've added it to the ToDo list. (I have vague memories of trying to update the version of cocotb before, running into problems, requesting support and not getting any. But I may be mis-remembering.)

imphil, 2021-03-26:
I think this can be closed -- cocotb 1.2 is available on edaplayground. I filed https://github.com/edaplayground/eda-playground/issues/46 to update to the latest 1.5 version.

closed 2021-03-29 by edaplayground