edaplayground/eda-playground #38: Not able to run codes in edaplayground.com, When clicking run page refreshes automatically

ojhatapaskumarr, 2020-07-24:
Hi Team,
My edaplayground.com account is registered with my college mail-id **m160006ec@nitsikkim.ac.in**. 
I am not able to run any code. When I am trying to run a code, the page automatically refreshes instead of giving any result. Also, I am not able to run my previous codes from my playground. The same issue is happening with these codes also.
Please help.

Thanks and regards,
Tapas

edaplayground, 2020-07-24:
Hi Tapas,

Please can you try logging in and out again; deleting all the cookies; a different browser - that sort of thing. Then let me know how you get on.

Matthew

ojhatapaskumarr, 2020-07-24:
Thank you Matthew for your timely reply. It got resolved after logging out and logging in again.

closed 2020-07-24 by ojhatapaskumarr