edaplayground/eda-playground #40: Support SystemC-AMS 2.3 and application examples

barnasc, 2020-11-17:
To support the creation of SystemC-based virtual prototypes including AMS and RF system functionality, it would be great if you can support SystemC-AMS. It can be downloaded free of charge here
https://www.coseda-tech.com/systemc-ams-proof-of-concept

Accellera released some SystemC-AMS application examples. Perhaps these can be added as examples. 
https://www.accellera.org/downloads/standards/systemc

Thanks Martin

edaplayground, 2020-11-24:
Thank you for this suggestion. I've added it to the ToDo list.

bardo84, 2024-12-04:
4 year later..

bardo84, 2024-12-05:
First, a big THANK YOU for maintaining this outstanding EdaPlayground site!

Adding the SystemCAMS includes and libraries follows exactly the same simple workflow as the already installed SystemC. 
See [https://systemc-ams.at/Setup.html](https://systemc-ams.at/Setup.html)
Above, Martin has mentioned the SystemC/AMS examples—there are numerous ones with a regression suite.
See [https://systemc-ams.at/Examples.html](https://systemc-ams.at/Examples.html)

What's in it for Doulos to showcase the only royalty-free analog mixed-signal simulator? 
I would guess increased popularity, leading to opportunities for SystemCAMS books and training courses.

I truly believe adding SystemCAMS to EdaPlayground is a low-hanging fruit.

barnasc, 2024-12-05:
let me chime in here to elaborate on the business aspect. 

We might face some sort of _chicken-egg_ problem here... If open platforms and sandboxes like this do not offer the full set of SystemC libraries and capabilities, adoption of these powerful extensions is more difficult.

Doulos did give a SystemC-AMS traning some decades ago, since I was involved in supplying some slide/training material. With the growing system complexity, in e.g. Automotive and IoT systems, the need for more abstract and efficient mixed signal system modeling concept is fundamental. SystemC-AMS is addressing this nicely, so it is up to Doulos to explore if there is a market opportunity here.

bardo84, 2024-12-05:
Thanks for reaching out!
Difficult to gauge the actual steps and effort involved on your side.
Here it is some simple make commands to compile and install.
The rest should stay the same, the user having to add some includes in the compile options.