edaplayground/eda-playground #49: Some of the saved playgrounds are not found on edaplayground

KLIKHITHA, 2021-04-26:


edaplayground, 2021-04-26:
Have you highlighted this option?
[image]

KLIKHITHA, 2021-04-26:
Yes..i did ..  Even after that i couldn't find

KLIKHITHA, 2021-04-26:
No..i have only one account and i use that only!

edaplayground, 2021-04-26:
Could you please tell me what email address you will have registered with and I can have a look and see what I can find.

edaplayground, 2021-04-26:
Or email getedaplayground@gmail.com, if you prefer. Then I'll see your email.

KLIKHITHA, 2021-04-26:
I have sent you the email address with which I have registered through email...Kindly check and resolve the issue!

edaplayground, 2021-04-26:
There are two accounts associated with the email you sent me. One is associated with a Google account with the same email; the other is associated with a Google account associated with the gmail address you sent it from.

KLIKHITHA, 2021-04-26:
So...what should i do now?

edaplayground, 2021-04-27:
You only have two playgrounds  - one in each account. Log in as one account and copy the code to (eg) a text editor. Log in as the other account and paste it back. Then both your playgrounds will be under one account.

KLIKHITHA, 2021-05-15:
Helloo...i couldn't get all my saved code! Can u help me? What should i do?

KLIKHITHA, 2021-05-16:
Helloo..how to resolve the issue
Unresolved hierarchial reference to intf.d_out from module "top" (module
not found) in the program
edaplayground.com/x/nit8
Please help me out!!!

edaplayground, 2021-05-17:
It looks like you've fixed this?

KLIKHITHA, 2021-05-17:
Yes!

KLIKHITHA, 2021-05-17:
Some of the saved playgrounds are not found on edaplayground

edaplayground, 2021-05-17:
I'm sorry - you are going to have to be more specific.

KLIKHITHA, 2021-05-17:
I didn't find some of my saved playground on edaplayground..u asked me to
login as one as i have two playgrounds ..to get all the playgrounds back!
How to do that?

edaplayground, 2021-05-17:
You need to log in with the email address you sent me as the username and whatever password goes with it. And you need to log in with the Google account that is associated with this email (by clicking on the Google icon on the login page). You need to take note of the URLs of the code in the account you don't want, log in with the account you do want, go to that URL and then click on "Copy".

KLIKHITHA, 2021-05-17:
Thank you, I will do that.