edaplayground/eda-playground #50: Mentor Questa vsim crashed

PauloCarvalhoSP, 2021-05-17:
When I running simulation UVM based, vsim command crashed and tool is killed.

See below the log:

[2021-05-14 12:40:22 EDT] vlib work && vlog -writetoplevels questa.tops '-timescale' '1ns/1ns' +incdir+$UVM_HOME/src -L $QUESTA_HOME/uvm-1.1d design.sv testbench.sv  && vsim -f questa.tops -batch -do run.do -voptargs=+acc=npr 
QuestaSim-64 vlog 2020.1_1 Compiler 2020.03 Mar  4 2020
Start time: 12:40:22 on May 14,2021
vlog -writetoplevels questa.tops -timescale 1ns/1ns "+incdir+/playground_lib/uvm-1.1d/src" -L /usr/share/questa/questasim//uvm-1.1d design.sv testbench.sv 
-- Compiling module tinyalu
-- Compiling module single_cycle
-- Compiling module three_cycle
-- Compiling package tinyalu_pkg
-- Importing package /usr/share/questa/questasim//uvm-1.1d.uvm_pkg (uvm-1.1d Built-in)
** Note: (vlog-2286) tinyalu_pkg.sv(18): Using implicit +incdir+/usr/share/questa/questasim/uvm-1.1d/../verilog_src/uvm-1.1d/src from import uvm_pkg
-- Compiling interface tinyalu_bfm
-- Importing package tinyalu_pkg
-- Compiling module top

Top level modules:
	top
End time: 12:40:23 on May 14,2021, Elapsed time: 0:00:01
Errors: 0, Warnings: 0
# vsim top -batch -do "run.do" -voptargs="+acc=npr" 
# Start time: 12:40:23 on May 14,2021
# ** Note: (vsim-3812) Design is being optimized...
# ** Warning: (vopt-10016) Option '-L /usr/share/questa/questasim//uvm-1.1d' was detected by vlog for design-unit 'top', but was not detected by vopt. The vlog option will be ignored.
# ** Warning: (vopt-10016) Option '-L /usr/share/questa/questasim//uvm-1.1d' was detected by vlog for design-unit 'tinyalu_pkg', but was not detected by vopt. The vlog option will be ignored.
# ** Warning: (vopt-10016) Option '-L /usr/share/questa/questasim//uvm-1.1d' was detected by vlog for design-unit 'tinyalu_bfm', but was not detected by vopt. The vlog option will be ignored.
# ** Warning: (vopt-10016) Option '-L /usr/share/questa/questasim//uvm-1.1d' was detected by vlog for design-unit 'tinyalu', but was not detected by vopt. The vlog option will be ignored.
# ** Warning: (vopt-10016) Option '-L /usr/share/questa/questasim//uvm-1.1d' was detected by vlog for design-unit 'single_cycle', but was not detected by vopt. The vlog option will be ignored.
# ** Warning: (vopt-10016) Option '-L /usr/share/questa/questasim//uvm-1.1d' was detected by vlog for design-unit 'three_cycle', but was not detected by vopt. The vlog option will be ignored.
# ** Note: (vsim-12126) Error and warning message counts have been restored: Errors=0, Warnings=6.
# //  Questa Sim-64
# //  Version 2020.1_1 linux_x86_64 Mar  4 2020
# //
# //  Copyright 1991-2020 Mentor Graphics Corporation
# //  All Rights Reserved.
# //
# //  QuestaSim and its associated documentation contain trade
# //  secrets and commercial or financial information that are the property of
# //  Mentor Graphics Corporation and are privileged, confidential,
# //  and exempt from disclosure under the Freedom of Information Act,
# //  5 U.S.C. Section 552. Furthermore, this information
# //  is prohibited from disclosure under the Trade Secrets Act,
# //  18 U.S.C. Section 1905.
# //
# Loading sv_std.std
# Loading work.tinyalu_bfm(fast)
# Loading /usr/share/questa/questasim//uvm-1.1d.uvm_pkg
# Loading work.tinyalu_pkg(fast)
# Loading /usr/share/questa/questasim//uvm-1.1d.questa_uvm_pkg(fast)
# Loading work.top(fast)
# Loading work.tinyalu_bfm(fast__2)
# Loading work.tinyalu(fast)
# Loading work.single_cycle(fast)
# Loading work.three_cycle(fast)
# Loading /usr/share/questa/questasim/uvm-1.1d/linux_x86_64/uvm_dpi.so
# 
# do run.do
# ** Warning: (vdel-134) Unable to remove locked optimized design "_opt".  Locker is unknown@fbd125514351.
# ** Warning: (vlib-34) Library already exists at "work".
# QuestaSim-64 vlog 2020.1_1 Compiler 2020.03 Mar  4 2020
# Start time: 12:40:29 on May 14,2021
# vlog "+incdir+/playground_lib/uvm-1.1d/src" -L /usr/share/questa/questasim/uvm-1.1d design.sv testbench.sv 
# -- Compiling module tinyalu
# -- Compiling module single_cycle
# -- Compiling module three_cycle
# -- Compiling package tinyalu_pkg
# -- Importing package /usr/share/questa/questasim/uvm-1.1d.uvm_pkg (uvm-1.1d Built-in)
# ** Note: (vlog-2286) tinyalu_pkg.sv(18): Using implicit +incdir+/usr/share/questa/questasim/uvm-1.1d/../verilog_src/uvm-1.1d/src from import uvm_pkg
# -- Compiling interface tinyalu_bfm
# -- Importing package tinyalu_pkg
# -- Compiling module top
# 
# Top level modules:
# 	top
# End time: 12:40:30 on May 14,2021, Elapsed time: 0:00:01
# Errors: 0, Warnings: 0
# QuestaSim-64 vopt 2020.1_1 Compiler 2020.03 Mar  4 2020
# Start time: 12:40:30 on May 14,2021
# vopt top -o top_optimized "+acc" "+cover=sbfec+tinyalu(rtl)." 
# 
# Top level modules:
# 	top
# 
# Analyzing design...
# ** Warning: (vopt-10016) Option '-L /usr/share/questa/questasim/uvm-1.1d' was detected by vlog for design-unit 'top', but was not detected by vopt. The vlog option will be ignored.
# -- Loading module top
# -- Importing package /usr/share/questa/questasim/uvm-1.1d.uvm_pkg (uvm-1.1d Built-in)
# ** Warning: (vopt-10016) Option '-L /usr/share/questa/questasim/uvm-1.1d' was detected by vlog for design-unit 'tinyalu_pkg', but was not detected by vopt. The vlog option will be ignored.
# -- Importing package tinyalu_pkg
# -- Loading package /usr/share/questa/questasim/uvm-1.1d.questa_uvm_pkg
# ** Warning: (vopt-10016) Option '-L /usr/share/questa/questasim/uvm-1.1d' was detected by vlog for design-unit 'tinyalu_bfm', but was not detected by vopt. The vlog option will be ignored.
# -- Loading interface tinyalu_bfm
# ** Warning: (vopt-10016) Option '-L /usr/share/questa/questasim/uvm-1.1d' was detected by vlog for design-unit 'tinyalu', but was not detected by vopt. The vlog option will be ignored.
# -- Loading module tinyalu
# ** Warning: (vopt-10016) Option '-L /usr/share/questa/questasim/uvm-1.1d' was detected by vlog for design-unit 'single_cycle', but was not detected by vopt. The vlog option will be ignored.
# -- Loading module single_cycle
# ** Warning: (vopt-10016) Option '-L /usr/share/questa/questasim/uvm-1.1d' was detected by vlog for design-unit 'three_cycle', but was not detected by vopt. The vlog option will be ignored.
# -- Loading module three_cycle
# Optimizing 9 design-units (inlining 0/5 module instances):
# -- Optimizing package /usr/share/questa/questasim/uvm-1.1d.uvm_pkg (uvm-1.1d Built-in)(fast)
# -- Optimizing package /usr/share/questa/questasim/uvm-1.1d.questa_uvm_pkg(fast)
# -- Optimizing package tinyalu_pkg(fast)
# -- Optimizing interface tinyalu_bfm(fast__2)
# -- Optimizing module three_cycle(fast)
# -- Optimizing module tinyalu(fast)
# -- Optimizing module single_cycle(fast)
# -- Optimizing module top(fast)
# -- Optimizing interface tinyalu_bfm(fast)
# Optimized design name is top_optimized
# End time: 12:40:35 on May 14,2021, Elapsed time: 0:00:05
# Errors: 0, Warnings: 6
# End time: 12:40:35 on May 14,2021, Elapsed time: 0:00:12
# Errors: 0, Warnings: 6
# vsim top_optimized -coverage "+UVM_TESTNAME=fibonacci_test" 
# Start time: 12:40:35 on May 14,2021
# ** Note: (vsim-12126) Error and warning message counts have been restored: Errors=0, Warnings=6.
# Attempting stack trace sig 11
# Signal caught: signo [11]
# vsim_stacktrace.vstf written
# Current time Fri May 14 12:40:35 2021
# Program = vsim
# Id = "2020.1_1"
# Version = "2020.03"
# Date = "Mar  4 2020"
# Platform = "linux_x86_64"
# Signature = 60deb3f7bea283b7f564c03ffc57ec3b
# 0    0x00000000015437a0: '<unknown (@0x15437a0)>'
# 1    0x00000000008dc3b0: '<unknown (@0x8dc3b0)>'
# 2    0x00000000009b80bc: '<unknown (@0x9b80bc)>'
# 3    0x00000000009bb6db: '<unknown (@0x9bb6db)>'
# 4    0x0000000000ea07a6: '<unknown (@0xea07a6)>'
# 5    0x0000000000e9a7fd: '<unknown (@0xe9a7fd)>'
# 6    0x000000000161603d: '<unknown (@0x161603d)>'
# 7    0x000000000161a496: '<unknown (@0x161a496)>'
# 8    0x000000000161bb81: '<unknown (@0x161bb81)>'
# 9    0x000000000161bee6: '<unknown (@0x161bee6)>'
# 10   0x000000000161d603: '<unknown (@0x161d603)>'
# 11   0x000000000161ddc1: '<unknown (@0x161ddc1)>'
# 12   0x0000000000b981c0: '<unknown (@0xb981c0)>'
# 13   0x0000000000b99c2d: '<unknown (@0xb99c2d)>'
# End of Stack Trace

** Fatal: (SIGSEGV) Bad pointer access. Closing vsimk.
** Fatal: vsimk is exiting with code 211.
Exit codes are defined in the "Error and Warning Messages"
appendix of the QuestaSim User's Manual.
# End time: 12:40:35 on May 14,2021, Elapsed time: 0:00:00
# Errors: 0, Warnings: 6
Exit code expected: 0, received: 1
Done

edaplayground, 2021-05-17:
This is a problem with Questa, not with EDA Playground. You need to comment things out until the crash goes away in order to narrow done what is causing the crash.

PauloCarvalhoSP, 2021-05-17:
The error occurs on this command:

vsim top_optimized -coverage +UVM_TESTNAME=fibonacci_test

The commands before this didn't crashed tool.

edaplayground, 2021-05-17:
As I say, it's a Mentor/Siemens issue, not an EDA Playground issue. You need to either use a different simulator or you need to do some good old-fashioned debugging to find out exactly which code is causing the crash and code it some other way if that is possible. You also might find that some other simulator might report an error (instead of crashing) which, if you fix it, might stop Questa crashing (because sometimes it is illegal code that causes simulators to crash rather than report the error).