edaplayground/eda-playground #57: Execution of file does not work

LK-14, 2021-11-27:
Hi there!

I was programming with VHDL today on the EDA playground, I wanted to execute my code (worked earlier) and without any warning or error just nothing happend and after 1min it timed out - does not work since. I also tested it with files which for sure work, the execution did not work there either. 
I asked a friend of mine and he had the same issue. Just wanted to know what I can do.

Thanks!

j-adie456980, 2021-11-27:
I am also having this issue running simulations for System Verilog.  Any info would be greatly appreciated.

Abinrose, 2021-11-28:
Im having same issue here..please help me if there any solutions available

mpataylor, 2021-11-29:
One of the servers was playing up. I have rebooted it and now everything seems to be OK. I'm sorry for the inconvenience.

closed 2021-11-29 by mpataylor