edaplayground/eda-playground #59: Unable to login using Facebook or Google Account

parth2488, 2021-12-16:
I used to login using my facebook account; but failed with below error.  Unable to login with google; failing with some error as well. 

In between jobs; so unable to reset the password to my account which I created using my company email. 

Please help me to login. Need help badly. I can provide my. login ID if required. 

**Facbook Login Error ==>** 
App Not Setup
This app is still in development mode, and you don't have access to it. Switch to a registered test user or ask an app admin for permissions.

**Google Login Error ==>**
Could not load the requested page
Error: javax.persistence.NonUniqueResultException: Should be only one user

Error Message: Should be only one user

javax.persistence.NonUniqueResultException: Should be only one user at com.victoreda.playground.domain.UserDAO.findByUsername(UserDAO.java:17) at com.victoreda.playground.security.LoggedInUserProvider.get(LoggedInUserProvider.java:20) at com.victoreda.playground.controllers.BaseController.getUser(BaseController.java:217) at com.victoreda.playground.controllers.HomeController.getHomePage(HomeController.java:327) at com.victoreda.playground.controllers.HomeController$$FastClassBySpringCGLIB$$df9fad65.invoke() at org.springframework.cglib.proxy.MethodProxy.invoke(MethodProxy.java:204) at org.springframework.aop.framework.CglibAopProxy$DynamicAdvisedInterceptor.intercept(CglibAopProxy.java:629) at com.victoreda.playground.controllers.HomeController$$EnhancerBySpringCGLIB$$1e24581b.getHomePage() at sun.reflect.GeneratedMethodAccessor313.invoke(Unknown Source) at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) at java.lang.reflect.Method.invoke(Method.java:606) at org.springframework.web.method.support.InvocableHandlerMethod.invoke(InvocableHandlerMethod.java:215) at org.springframework.web.method.support.InvocableHandlerMethod.invokeForRe

mpataylor, 2021-12-16:
Facebook has been broken for a while. I am struggling to fix it, sorry. I know what the problem is with your Google login - some people have duplicate entries in the user database. I am just about to start teaching, so i can't sort it out now.

mpataylor, 2021-12-16:
Actually, you'll need to email getedaplayground@gmail.com from the email address you have registered with Google so that I know what to look for in the database.

parth2488, 2021-12-16:
Just emailed; but if you want I can add here as well