edaplayground/eda-playground #62: Problem connecting to server

Lobo2367, 2022-03-18:
Tipo Informe de Excepción

mensaje javax.el.ELException: Error reading [siteMessageNoHtml] on type [com.victoreda.playground.controllers.HomeController$$EnhancerBySpringCGLIB$$b87acceb]

Descripción El servidor encontró un error interno que hizo que no pudiera rellenar este requerimiento.

excepción

org.apache.jasper.JasperException: javax.el.ELException: Error reading [siteMessageNoHtml] on type [com.victoreda.playground.controllers.HomeController$$EnhancerBySpringCGLIB$$b87acceb]
	org.apache.jasper.servlet.JspServletWrapper.handleJspException(JspServletWrapper.java:639)
	org.apache.jasper.servlet.JspServletWrapper.service(JspServletWrapper.java:515)

mpataylor, 2022-03-18:
Please could you give more information. What are you doing to cause this error?

Lobo2367, 2022-03-18:
Its already fixed