edaplayground/eda-playground #63: EPWave doesn't work showing "edaplayground.com refused to connect" message instead of the waves

tolaszi, 2022-03-28:
When I run a simulation with the "Open EPWave after run" option enabled the EPWave window pops up but it shows the above mentioned "refused to connect message". I am logged in, see the same behavior with all four commercial simulators and both with VHDL and Verilog designs (with dumpfile code snippet added).

mpataylor, 2022-03-28:
Sorry you're having a problem opening EPWave. Please could you change the state of this option (on your profile page):

[image]

and save it. And then tell me whether this has any affect?

mpataylor, 2022-03-28:
Please can you tell me the URL of the code you are trying to run?

tolaszi, 2022-03-28:
I don't save my codes onto EDA playground but I have tried with a bunch of the published examples with the same result.

Even the most basic one at https://edaplayground.com/x/9 gives me the same result.

This time I have seen a red message in the console log as well:
```
Pipe broken
Execution interrupted or reached maximum runtime.
```

tolaszi, 2022-03-29:
With the new day the issue seems to have been gone. I can now see the waveforms appearing with all the simulators. 

Thank you very much for your help on this! Please feel free to close this issue if you see the same on your side.

mpataylor, 2022-03-29:
Glad to hear it's disappeared. I didn't do anything to fix it. Yet. In fact, I'm working on rolling the whole site back one version, because this issue has appeared since I updated it. (It's an intermittent issue: I didn't find it when testing the new release.) Don't worry - there is a second intermittent issue that has appeared since the update - your reported issue was the straw that broke the camel's back. I'll (try to) fix both these issues offline before rolling everything forward again.

closed 2022-03-29 by mpataylor