edaplayground/eda-playground #66: EDA Playground web page is always loading, unable to show the editor page

guohaibo55, 2022-05-20:
Starting from May 19, the main page of https://www.edaplayground.com/ failed to load, showing a large "Loading..." on a gray background. The status line showed a message that it was trying to connect to a website called cdn.jsdelivr.net. The website cdn.jsdelivr.net had temporally down for at least two times in the past two years in mainland China, from where I was trying to connect to EDA Playground. It appears that the cdn site somehow stopped EDA Playground from correctly loading, but I am not sure about this. Is there a solution for this?

lewis1573, 2022-05-25:
Reason is GFW killed jsdelivr in China, https://luotianyi.vc/6295.html

mpataylor, 2022-05-25:
I did once try to get EDA Playground to work reliably in China. I discovered that this is very difficult because censorship in China is not controlled at a national level but at a local level. So, there is no way to find out what is and what is not censored and no way to test it.

mpataylor, 2022-05-25:
Sorry lewis, I don't understand your post.

guohaibo55, 2022-05-25:
Thank you for explaining this. It appears there are no solutions to this at this time. This is really inconvenient as some of my students prefer to use EDA Playground with their tablets or even smart phones. They simply do not like to use computers. EDA Playground is an excellent online editor and simulator, and I will come back from time to time to see if the problem is gone.

lewis1573, 2022-06-08:
Here's a solution, https://github.com/PipecraftNet/jsdelivr-auto-fallback
For regions that cannot access cdn.jsdelivr.net, automatically switch to another accessible cdn, such as fastly.jsdelivr.net, and for regions that can access cdn.jsdelivr.net, continue to use cdn.jsdelivr.net, which seems to require very little code changes. Can the maintainer consider adding this feature? Many thanks!

closed 2024-09-12 by edaplayground