edaplayground/eda-playground #75: Same issue of Max runtime and Error connecting to server. Run request failed: error

Inzendis, 2022-10-11:
About 2 days ago it was working just fine with the same vhdl code. The testbench only lasts 140 ns and was running without problems. As of this morning around 2am EST and now 4:25pm EST its still doing the same thing. I logged out and logged in and nothing, also it gave me a weird error page when logging in twice. HTTP Status 404

Inzendis, 2022-10-11:
[image]

Inzendis, 2022-10-11:
Tested in Google Chrome and IE with the same outcome. Yet tested in firefox with a virtual machine linux ubuntu and it was working just fine. Maybe its my computer.

Inzendis, 2022-10-11:
It did work on my google chrome 3 times in a row, until it happened again just now.

edaplayground, 2022-10-11:
It is weird and inconsistent. I don't know what causes it. You need to reload the page. If you have code to save, you'll have to copy it (and perhaps paste it somewhere else).