edaplayground/eda-playground #79: Unable to run any simulation

itsnishit, 2023-04-09:
Whenever I try to save the code it shows this :
[image]
Whenever I try to run the code it shows this :
[image]

kulkarniakshay02, 2023-04-13:
I am also getting the same error when I try to run specman e language related tests:

[image]

edaplayground, 2023-04-17:
Did you try reloading the page?

kulkarniakshay02, 2023-04-17:
Yes yes, tried reloading the page but, I get this error more often than the
times it works.

Akshay

Daniyal-Tahsildar, 2023-04-19:
I have been facing the same issue as well. Crashes more times that it successfully runs. Kindly let me know how to resolve this. By the way, I have been working using UVM.

edaplayground, 2023-04-19:
The problem is a bug. It tends to happen when you stay connected to a page and try to run (or do anything else) without first refreshing the page. I don't know any more than that. When I have time to look at it, I will be able to try to fix it. But I never have the time. Sorry for the inconvenience.

edaplayground, 2023-04-20:
I'm afraid this is a bug. I don't know what causes it. I have not had the
time to look into it. I think it is worse at busy times of the day (the
working day in India). You could try reloading the page or waiting until a
quieter time.

Matthew