edaplayground/eda-playground #81: Unable to register with Google account

dani1024, 2023-05-02:
I'm trying to register with my google account but I get an HTTP 500 error

HTTP Status 500 - PreparedStatementCallback; SQL [insert into UserConnection (userId, providerId, providerUserId, rank, displayName, profileUrl, imageUrl, accessToken, secret, refreshToken, expireTime) values (?, ?, ?, ?, ?, ?, ?, ?, ?, ?, ?)]; Data truncation: Data too long for column 'imageUrl' at row 1; nested exception is com.mysql.jdbc.MysqlDataTruncation: Data truncation: Data too long for column 'imageUrl' at row 1

Can you please check that?
Thank you

edaplayground, 2023-05-03:
Please could you tell me what email address the Google account is associated with. If you don't want to post it here, please email getedaplayground@gmail.com .

dani1024, 2023-05-10:
Hi, I sent my gmail account last week by email. Did you get it? 
Thank you

edaplayground, 2023-05-11:
Hi Daniel,

> Hi, I sent my gmail account last week by email. Did you get it?

Well, I did. But for some reason, I missed it. Sorry.

So, the problem was clear. For some reason, you have 17 identical accounts
registered. Which is odd, because you ought to be only able to create one
per email address. Weird. Anyway, I have deleted 16 of them,
fingers-crossed that sorts it.

Best regards,

Matthew

dani1024, 2023-05-11:
Hi Matthew

No luck... same error again.
Perhaps when clicking on "login" it tries to create an account again instead of loging into it.
Can you try deleting all the accounts? (I guess there are 2 now)

Many thanks

edaplayground, 2023-05-11:
I've deleted both accounts. Please try again.

Matthew

edaplayground, 2023-05-11:
(You were right - there were two accounts. Weird.)

dani1024, 2023-05-11:
Still nothing, same error when creating the account for the first time. 
Don't want to bother you... I can use my corporate email to create a new account (although this is for hobbyist use)
Thank you for the help,

Daniel