edaplayground/eda-playground #83: HTTP Status 500 – Internal Server Error

Thirum6, 2023-08-19:
As of now, not able to use EDA playground. 
Its Giving error "HTTP Status 500 – Internal Server Error". 
I restarted my system and lunched again with new browsers, still facing the same issue. 
Can anyone give solution?

rameshchevuru, 2024-07-16:
Hi 
Iam still getting http status 500 internal Server Error after deleted cookies and history in any browser.
Iam notice that its issue with my register mail. If i was login with another mail ID then able to access EDA PLAYGROUND.
please Suggest to this issue

edaplayground, 2024-07-16:
What is you email address, please? If you don't want to post it here, please email support@edaplayground.com .

rameshchevuru, 2024-07-16:
***@***.***

edaplayground, 2024-07-16:
It looks like GitHub censors email addresses. You can find the support address on the EDA Playground page.

edaplayground, 2024-07-17:
I need to know your email address. That is your username on EDA Playground. I cannot do anything without knowing that.

rameshchevuru, 2024-07-17:
***@***.***

edaplayground, 2024-07-17:
But, as I've said already, GitHub is censoring email addresses. Please forward it to the EDA playground support address, which you can find via the EDA Playground website. (I can't post it here, because GitHub will censor it.)

[image]

rameshchevuru, 2024-07-18:
Thanks for the tip!