edaplayground/eda-playground #87: Redirecting me to log-in page everytime I save a file

butterr12, 2024-12-08:
Hello! 

As the title suggests, how do I resolve the issue of EDA playground redirecting me to log-in page every time I save a file? Thank you.

butterr12, 2024-12-21:
Hello, this has been a persistent issue for me. The number is 209; I would appreciate any help. Thank you!

edaplayground, 2024-12-23:
I tested 209: it's working fine. (TBH, had there been a problem with it for two whole weeks, I'd have been hearing a lot more about it.) The next thing is to look at your account. I need to know the email address you registered with to do that. (Or the email address the Google account is associated with if that's how you logged in.) Please can you email: support@edaplayground.com from that address and mark it for my (Matthew's) attention?

Harika082324, 2025-08-21:
Redirecting me to log-in page everytime I save a file how do I resolve the issue of EDA playground redirecting me to log-in page or HTTP Status 403 – Forbidden every time I save a file?plz how to resolve this issue

Harika082324, 2025-08-21:
> I'm sorry you're having difficulties. I assume you've tried the things
> suggested above. Please could you tell me the ID of the server you are
> using (also as suggested above).

***@***.***
This is my Id,plz solve this problem

Thank you

edaplayground, 2025-08-21:
Thank you. There's nothing wrong with that particular server, as far as I can see. Github has filtered out your email address. If you could tell me your name or enough of your email to allow me to single you out in the database, that would be good. Searching for 'harika' returns 449 hits!

Harika082324, 2025-08-21:
Google it is and I have done all of them

Harika082324, 2025-08-21:
Ok

edaplayground, 2025-08-21:
Thank you. I can't see anything wrong with your account. Have you got any add-ins on your browser like ad blockers and suchlike? If so, please can you try turning them off? Or try a different browser?

Harika082324, 2025-08-21:
No.Actually I have done all the above things.and i am not able to login also .what can I do now ?

Harika082324, 2025-08-21:
I am giving my credentials to login but again it was redirecting to the same login page and trying for so many time ,if I get the editor,then if I save my code then it was redirecting to login or error page .this is issue I am getting

edaplayground, 2025-08-21:
OK. I've found that. It all looks OK. When I try to log in using that, I get "Invalid password", which is what I would expect because I don't know your password. (It is encrypted.) What happens if you try to log in with the wrong password?

Harika082324, 2025-08-21:
It was coming like captcha validation failed.please try again

Harika082324, 2025-08-21:
If I try wrong password I am getting the captcha invalid please try again.As I mentioned before ,if I try for so many times then only I am able to login

edaplayground, 2025-08-21:
I don't know then. It just works for me: I can't reproduce any problem. Are you using a VPN? Can you bypass that?

edaplayground, 2025-08-21:
I need to go now. Try again tomorrow if it's still not working.

Harika082324, 2025-08-21:
Ok .I will let you know