edaplayground/eda-playground #88: Parse of do command

skyviper, 2025-02-19:
When I run 2 word do commands for the questa tool, the command is parsed incorrectly.
A -do run 0 causes an error.
A -do uvm printfactory  also causes an error.

skyviper, 2025-02-20:
>   How about
> -do "run 0"
> -do "uvm printfactory"
>
> How about
> -do "run 0"
> -do "uvm printfactory"
> ?

That doesn’t work

edaplayground, 2025-02-21:
How about putting those commands in a run.do file?

https://www.edaplayground.com/x/sMRX

skyviper, 2025-02-21:
I ran mine with the run.do as you did in your example and the factory doesn't print.

[image]

skyviper, 2025-02-22:
I solved the issue. My test bench wasn't taking any time. So there wasn't any time to process the requests.

closed 2025-02-22 by skyviper